ros-drivers/flir_camera_driver #224: [Jazzy] Fails to connect to Grasshopper USB3 on WSL Ubuntu 24.04 with

PeteBlackerThe3rd, 2024-12-13:
**Is this behavior also observed when using SpinVIew?**
We have SpinView setup on an Ubuntu 22.04 WSL VM on the same machine and it's working fine. SpinView doesn't appear to support Ubuntu 24.04, we've tried to install it but it's compiled against older libraries and fails to run.

**Describe the bug**
Initially installed ros-jazzy-spinnaker-camera-driver using apt, and completed installation steps. The default frame-rate needed to be reduced from 100 to 30, after that all configuration parameters were updated correctly. Then the node crashes with: 
```terminate called after throwing an instance of 'Spinnaker::Exception'
what():  Spinnaker: Could not start acquisition. Please try reconnecting the device. [-1010]```

We've built the driver from source to try and identify where this exception is thrown. We've built 'humble-devel' 'humble-release' and 'ros2-release' All of these branches fail to set any of the camera parameters, even though the Debian package installed version succeed here.

We've narrowed down the source of the exception to spinnaker_wrapper_impl.cpp and the set_acquisition_mode_continuous() function when it is called by SpinnakerWrapperImpl::startCamera

If you could let us know which point in the codebase we need to build to get identical behavior to the Debian packaged version we could continue debugging from here.

**Useful logging output**
Console output when running debian package installed verison:
```
$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=grasshopper serial:="'22415660'"
[INFO] [launch]: All log files can be found below ------
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [11628]
[camera_driver_node-1] [INFO] [1734086532.785963064] [22415660]: reading ros parameters for camera with serial: 22415660
[camera_driver_node-1] [INFO] [1734086532.786021228] [22415660]: not adjusting time stamps!
[camera_driver_node-1] [INFO] [1734086532.786052676] [22415660]: parameter definitions file: /opt/ros/jazzy/share/spinnaker_camera_driver/config/grasshopper.yaml
[camera_driver_node-1] [INFO] [1734086532.791310840] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1734086532.791363332] [flir_camera]: camera calibration URL: file:///home/pete/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [INFO] [1734086533.834352174] [22415660]: using spinnaker lib version: 4.0.0.116
[camera_driver_node-1] [INFO] [1734086533.834708895] [22415660]: found camera with serial number: 22415660
[camera_driver_node-1] [INFO] [1734086534.130849470] [22415660]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] [INFO] [1734086534.138087939] [22415660]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] [INFO] [1734086534.148790629] [22415660]: setting AcquisitionControl/AcquisitionFrameRateAuto to: Off
[camera_driver_node-1] [INFO] [1734086534.155266723] [22415660]: setting AcquisitionControl/AcquisitionFrameRate to: 30
[camera_driver_node-1] [INFO] [1734086534.164425589] [22415660]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1734086534.174008761] [22415660]: setting ChunkDataControl/ChunkModeActive to: 1
[camera_driver_node-1] [INFO] [1734086534.179369043] [22415660]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[camera_driver_node-1] [INFO] [1734086534.184912093] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1734086534.196011401] [22415660]: setting ChunkDataControl/ChunkSelector to: Gain
[camera_driver_node-1] [INFO] [1734086534.199726988] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1734086534.206759401] [22415660]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] [INFO] [1734086534.210746451] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1]   what():  Spinnaker: Could not start acquisition. Please try reconnecting the device. [-1010]
[ERROR] [camera_driver_node-1]: process has died [pid 11628, exit code -6, cmd '/opt/ros/jazzy/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r __node:=flir_camera --params-file /tmp/launch_params_ahfwup1j --params-file /tmp/launch_params_ktkflpv_ -r ~/control:=/exposure_control/control'].
```

Console output when built from source from ros2-release branch:
```
$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=grasshopper serial:="'22415660'"
[INFO] [launch]: All log files can be found below --------
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [11541]
[camera_driver_node-1] [INFO] [1734086074.526915221] [22415660]: reading ros parameters for camera with serial: 22415660
[camera_driver_node-1] [INFO] [1734086074.526969740] [22415660]: not adjusting time stamps!
[camera_driver_node-1] [INFO] [1734086074.527003655] [22415660]: parameter definitions file: /home/pete/flir_camera_driver/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/grasshopper.yaml
[camera_driver_node-1] [INFO] [1734086074.529106717] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1734086074.529161812] [flir_camera]: camera calibration URL: file:///home/pete/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [INFO] [1734086075.565260727] [22415660]: using spinnaker lib version: 4.0.0.116
[camera_driver_node-1] [INFO] [1734086075.565566967] [22415660]: found camera with serial number: 22415660
[camera_driver_node-1] [INFO] [1734086075.914411701] [Spinnaker Wrapper]: Initialized camera [serial: 22415660] from: [USB Interface 0]
[camera_driver_node-1] [INFO] [1734086075.914508199] [22415660]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] no category node: Root vs AnalogControl/GainAuto
[camera_driver_node-1] no category node: Root vs AnalogControl/GainAuto
[camera_driver_node-1] no category node: Root vs AnalogControl/GainAuto
[camera_driver_node-1] [WARN] [1734086075.914551927] [22415660]: setting AnalogControl/GainAuto failed: node AnalogControl/GainAuto not found!
[camera_driver_node-1] [WARN] [1734086075.914553789] [22415660]: AnalogControl/GainAuto set to: UNKNOWN instead of: Continuous
[camera_driver_node-1] [INFO] [1734086075.914651925] [22415660]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] no category node: Root vs AcquisitionControl/ExposureAuto
[camera_driver_node-1] no category node: Root vs AcquisitionControl/ExposureAuto
[camera_driver_node-1] no category node: Root vs AcquisitionControl/ExposureAuto
[camera_driver_node-1] [WARN] [1734086075.914660579] [22415660]: setting AcquisitionControl/ExposureAuto failed: node AcquisitionControl/ExposureAuto not found!
[... 41 lines cut ...]
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkEnable
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkEnable
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkEnable
[camera_driver_node-1] [WARN] [1734086075.914807774] [22415660]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable not found!
[camera_driver_node-1] [INFO] [1734086075.914813094] [22415660]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkSelector
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkSelector
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkSelector
[camera_driver_node-1] [WARN] [1734086075.914819424] [22415660]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector not found!
[camera_driver_node-1] [WARN] [1734086075.914821087] [22415660]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Timestamp
[camera_driver_node-1] [INFO] [1734086075.914849949] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkEnable
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkEnable
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkEnable
[... cut ...] unch_params_y2oq2m9w --params-file /tmp/launch_params_nb1k_c8t -r ~/control:=/exposure_control/control'].
```

**System details**
- operating system/version: Ubuntu 24.04 running within WSL2 on Windows 11 
- USB devices forwarded to WSL2 over usbipd as described in this guide:  https://learn.microsoft.com/en-us/windows/wsl/connect-usb
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]): Jazzy
- complete FLIR camera model (e.g. BFS-U3-04S2C-C): Grasshopper GS3-U3-41C6C

berndpfrommer, 2024-12-13:
I have not seen this one before.
You can try using the humble-devel branch and roll back to commit 30c8e0c51faef9403f54a66b2e41d0bc51999c4f
in case I messed something up in a recent commit.
The disconnect looks weird though, and the nodes not found as well. Could be a WSL issue. I have never used WSL.

What does lsusb -t tell you? Does the cam show as USB3 at 5000 Mbits/s?
Also, does the cam work fine using spinview on the same host under windows?
Have you tried rebooting the host and/or replugging the camera?

PeteBlackerThe3rd, 2024-12-13:
Thanks for the quick reply!

I've pulled commit 30c8e0c and build it. It's still producing the warnings when it fails to set the parameters like the other branches.  I've updated the Grasshopper config with  'dump_node_map': True, this prints "dumping node map!" but no more to the console. Extra log info has been added to the SpinnakerWrapperImpl::startCamera function to identify that the exception is thrown in the call to set_acquisition_mode_continuous(). Is it possible it's not getting a valid nodeMap at all for some reason?

```
$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=grasshopper serial:="'22415660'"
[INFO] [launch]: All log files can be found below ---
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [1795]
[camera_driver_node-1] [INFO] [1734098023.711372797] [22415660]: reading ros parameters for camera with serial: 22415660
[camera_driver_node-1] [INFO] [1734098023.711419650] [22415660]: not adjusting time stamps!
[camera_driver_node-1] [INFO] [1734098023.711451101] [22415660]: parameter definitions file: /home/pete/flir_camera_driver/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/grasshopper.yaml
[camera_driver_node-1] [INFO] [1734098023.712731497] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1734098023.712779085] [flir_camera]: camera calibration URL: file:///home/pete/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [INFO] [1734098024.814351505] [22415660]: using spinnaker lib version: 4.0.0.116
[camera_driver_node-1] [INFO] [1734098024.814707356] [22415660]: found camera with serial number: 22415660
[camera_driver_node-1] [INFO] [1734098025.063379001] [Spinnaker Wrapper]: Initialized camera [serial: 22415660] from: [USB Interface 0]
[camera_driver_node-1] [INFO] [1734098025.063423718] [22415660]: dumping node map!
[camera_driver_node-1] [INFO] [1734098025.063468362] [22415660]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] no category node: Root vs AnalogControl/GainAuto
[camera_driver_node-1] no category node: Root vs AnalogControl/GainAuto
[camera_driver_node-1] no category node: Root vs AnalogControl/GainAuto
[camera_driver_node-1] [WARN] [1734098025.063488842] [22415660]: setting AnalogControl/GainAuto failed: node AnalogControl/GainAuto not found!
[camera_driver_node-1] [WARN] [1734098025.063490435] [22415660]: AnalogControl/GainAuto set to: UNKNOWN instead of: Continuous
[camera_driver_node-1] [INFO] [1734098025.063564904] [22415660]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] no category node: Root vs AcquisitionControl/ExposureAuto
[camera_driver_node-1] no category node: Root vs AcquisitionControl/ExposureAuto
[camera_driver_node-1] no category node: Root vs AcquisitionControl/ExposureAuto
[camera_driver_node-1] [WARN] [1734098025.063570945] [22415660]: setting AcquisitionControl/ExposureAuto failed: node AcquisitionControl/ExposureAuto not found!
[camera_driver_node-1] [WARN] [1734098025.063572629] [22415660]: AcquisitionControl/ExposureAuto set to: UNKNOWN instead of: Continuous
[camera_driver_node-1] [INFO] [1734098025.063578981] [22415660]: setting AcquisitionControl/AcquisitionFrameRateAuto to: Off
[camera_driver_node-1] no category node: Root vs AcquisitionControl/AcquisitionFrameRateAuto
[camera_driver_node-1] no category node: Root vs AcquisitionControl/AcquisitionFrameRateAuto
[camera_driver_node-1] no category node: Root vs AcquisitionControl/AcquisitionFrameRateAuto
[camera_driver_node-1] [WARN] [1734098025.063584125] [22415660]: setting AcquisitionControl/AcquisitionFrameRateAuto failed: node AcquisitionControl/AcquisitionFrameRateAuto not found!
[camera_driver_node-1] [WARN] [1734098025.063585608] [22415660]: AcquisitionControl/AcquisitionFrameRateAuto set to: UNKNOWN instead of: Off
[camera_driver_node-1] [INFO] [1734098025.063597820] [22415660]: setting AcquisitionControl/AcquisitionFrameRate to: 30
[camera_driver_node-1] no category node: Root vs AcquisitionControl/AcquisitionFrameRate
[camera_driver_node-1] no category node: Root vs AcquisitionControl/AcquisitionFrameRate
[camera_driver_node-1] no category node: Root vs AcquisitionControl/AcquisitionFrameRate
[camera_driver_node-1] [WARN] [1734098025.063604488] [22415660]: setting AcquisitionControl/AcquisitionFrameRate failed: node AcquisitionControl/AcquisitionFrameRate not found!
[camera_driver_node-1] [INFO] [1734098025.063613682] [22415660]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] no category node: Root vs AcquisitionControl/TriggerMode
[camera_driver_node-1] no category node: Root vs AcquisitionControl/TriggerMode
[camera_driver_node-1] no category node: Root vs AcquisitionControl/TriggerMode
[camera_driver_node-1] [WARN] [1734098025.063618759] [22415660]: setting AcquisitionControl/TriggerMode failed: node AcquisitionControl/TriggerMode not found!
[camera_driver_node-1] [WARN] [1734098025.063620093] [22415660]: AcquisitionControl/TriggerMode set to: UNKNOWN instead of: Off
[camera_driver_node-1] [INFO] [1734098025.063632908] [22415660]: setting ChunkDataControl/ChunkModeActive to: 1
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkModeActive
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkModeActive
[camera_driver_node-1] no category node: Root vs ChunkDataControl/ChunkModeActive
[... cut ...] ile /tmp/launch_params_4pn_5xh9 -r ~/control:=/exposure_control/control'].
```

Once the USB camera is attached to WSL using usbipd, in Ubuntu I see this:
```
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 1e10:3300 Point Grey Research, Inc. Grasshopper3 GS3-U3-41C6C
```

SpinView in Windows detected and captures from the camera without issue.

I've tried several reboots and disconnects.

PeteBlackerThe3rd, 2024-12-13:
Verbose lsusb output:

```
Bus 002 Device 002: ID 1e10:3300 Point Grey Research, Inc. Grasshopper3 GS3-U3-41C6C
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 [unknown]
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1e10 Point Grey Research, Inc.
  idProduct          0x3300 Grasshopper3 GS3-U3-41C6C
  bcdDevice            0.00
  iManufacturer           1 Point Grey Research
  iProduct                2 Grasshopper3 GS3-U3-41C6C
  iSerial                 3 0156092C
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0074
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0
      iFunction               4 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0
      iInterface              0
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 05 06 07 08 09 0a 0b 0c 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      1
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      2
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f61e
      BESL Link Power Management (LPM) Supported
    BESL value     1536 us
    Deep BESL value    61440 us
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000c
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   2
      Lowest fully-functional device speed is High Speed (480Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         512 micro seconds
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled
```

berndpfrommer, 2024-12-13:
I was really looking for "lsusb -t". The -t shows a tree, with the speed of the devices. 

My gut tells me this is not a bug in the ROS2 driver, but a system or permission issue. The question is also why the different behavior with the system packages vs compiled from source? Are the packages treated differently permissions wise than random code run by the user?

Is running ROS2 humble on the 22.04 machine an option? Not sure what to do if that one fails as well. I believe SpinView uses the Spinakker SDK underneath, so the hope is if SpinView works, so should the ROS2 driver.

Also check the system log (/var/log/syslog and dmesg on Ubuntu). Maybe some clue there as to what the problem is.

PeteBlackerThe3rd, 2024-12-13:
I wasn't planning on setting up ROS on the 22.04 machine, but right now it's looking like a useful option! I'll get it installed and let you know what I find. 

Here are lsusb outputs:

```
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 004: ID 1e10:3300 Point Grey Research, Inc. Grasshopper3 GS3-U3-41C6C

$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=vhci_hcd/8p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=vhci_hcd/8p, 5000M
    |__ Port 001: Dev 004, If 0, Class=Miscellaneous Device, Driver=[none], 5000M
    |__ Port 001: Dev 004, If 1, Class=Miscellaneous Device, Driver=[none], 5000M
    |__ Port 001: Dev 004, If 2, Class=Miscellaneous Device, Driver=[none], 5000M
```

Here is the output of dmesg, I think covering the failed node startup (syslog contains the same messages as far as I can tell):
```
[14074.509109] usb 2-1: SetAddress Request (4) to port 0
[14074.509121] usb 2-1: reset SuperSpeed USB device number 4 using vhci_hcd
[14074.637300] potentially unexpected fatal signal 6.
[14074.637304] CPU: 11 PID: 3952 Comm: camera_driver_n Not tainted 5.15.167.4-microsoft-standard-WSL2 #1
[14074.637307] RIP: 0033:0x7f7c6531db1c
[14074.637311] Code: 00 f0 ff ff 76 8f 41 89 c6 41 f7 de eb 8a b8 ba 00 00 00 0f 05 89 c3 e8 82 6f 05 00 44 89 e2 89 de 89 c7 b8 ea 00 00 00 0f 05 <41> 89 c6 41 f7 de 3d 00 f0 ff ff b8 00 00 00 00 44 0f 46 f0 e9 7b
[14074.637313] RSP: 002b:00007fff9094ea40 EFLAGS: 00000246 ORIG_RAX: 00000000000000ea
[14074.637316] RAX: 0000000000000000 RBX: 0000000000000944 RCX: 00007f7c6531db1c
[14074.637318] RDX: 0000000000000006 RSI: 0000000000000944 RDI: 0000000000000944
[14074.637318] RBP: 00007fff9094ea80 R08: 0000000000000000 R09: 00007fff9094e5d0
[14074.637319] R10: 0000000000000008 R11: 0000000000000246 R12: 0000000000000006
[14074.637320] R13: 0000561b51abd820 R14: 0000000000000016 R15: 00007f7c64660066
[14074.637322] FS:  00007f7c61ac3940 GS:  0000000000000000
[14074.639492] WSL (4034): Capturing crash for pid: 2372, executable: !home!pete!flir_camera_driver!build!spinnaker_camera_driver!camera_driver_node
[14074.639495] , signal: 6, port: 50005
[14078.151078] vhci_hcd: unlink->seqnum 22650
[14078.151088] vhci_hcd: urb->status -104
[14078.151230] vhci_hcd: unlink->seqnum 22651
[14078.151237] vhci_hcd: urb->status -104
[14078.151245] vhci_hcd: unlink->seqnum 22652
[14078.151246] vhci_hcd: urb->status -104
[14078.151247] vhci_hcd: unlink->seqnum 22653
[14078.151248] vhci_hcd: urb->status -104
[14078.151325] vhci_hcd: unlink->seqnum 22654
[14078.151327] vhci_hcd: urb->status -104
[14078.151345] vhci_hcd: unlink->seqnum 22655
[14078.151346] vhci_hcd: urb->status -104
[14078.151347] vhci_hcd: unlink->seqnum 22656
[14078.151348] vhci_hcd: urb->status -104
[14078.151370] vhci_hcd: unlink->seqnum 22657
```

PeteBlackerThe3rd, 2024-12-14:
After installing ROS2 humble and the spinnaker_camera_driver using apt on the 22.04 machine, the node is still failing to start the camera. If I run spinview in the same console immediately after this it detects and opens the camera without any problems.

I've also tried changing the owner and group of the install folder when building the driver from source to root:root to match that of the Debian package installed version. This hasn't affects the failure.

Do you have any idea what commit of the repo has been build into the debian packages distributed via apt? It would be really useful to get the same behavior built from source?

I should be able to dig a lot more into the nodemap issue that seems to be causing the problem, but that will probably have to wait until next week now. 

Thanks for your help so far! 

```
/opt/ros/humble/share/spinnaker_camera_driver/launch$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=grasshopper serial:="'22415660'"
[INFO] [launch]: All log files can be found below ---
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [16522]
[camera_driver_node-1] [INFO] [1734133687.565180530] [22415660]: reading ros parameters for camera with serial: 22415660
[camera_driver_node-1] [INFO] [1734133687.565255362] [22415660]: not adjusting time stamps!
[camera_driver_node-1] [INFO] [1734133687.565304152] [22415660]: parameter definitions file: /opt/ros/humble/share/spinnaker_camera_driver/config/grasshopper.yaml
[camera_driver_node-1] [INFO] [1734133687.573405036] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1734133687.573474575] [flir_camera]: camera calibration URL: file:///home/pete/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [INFO] [1734133688.958797321] [22415660]: using spinnaker lib version: 3.1.0.79
[camera_driver_node-1] [INFO] [1734133688.959081177] [22415660]: found camera with serial number: 22415660
[camera_driver_node-1] [INFO] [1734133689.103354469] [Spinnaker Wrapper]: Initialized camera [serial: 22415660] from: [USB Interface 0]
[camera_driver_node-1] [INFO] [1734133689.103446937] [22415660]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] [INFO] [1734133689.107167677] [22415660]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] [INFO] [1734133689.110724635] [22415660]: setting AcquisitionControl/AcquisitionFrameRateAuto to: Off
[camera_driver_node-1] [INFO] [1734133689.114334608] [22415660]: setting AcquisitionControl/AcquisitionFrameRate to: 30
[camera_driver_node-1] [INFO] [1734133689.118508425] [22415660]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1734133689.120815791] [22415660]: setting ChunkDataControl/ChunkModeActive to: 1
[camera_driver_node-1] [INFO] [1734133689.123220631] [22415660]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[camera_driver_node-1] [INFO] [1734133689.125328771] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1734133689.127625768] [22415660]: setting ChunkDataControl/ChunkSelector to: Gain
[camera_driver_node-1] [INFO] [1734133689.129238951] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1734133689.131565935] [22415660]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] [INFO] [1734133689.132792391] [22415660]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1]   what():  Spinnaker: Could not start acquisition. Please try reconnecting the device. [-1010]
[ERROR] [camera_driver_node-1]: process has died [pid 16522, exit code -6, cmd '/opt/ros/humble/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r __node:=flir_camera --params-file /tmp/launch_params_zxhage5r --params-file /tmp/launch_params_ixcq0kl4 -r ~/control:=/exposure_control/control'].
```

berndpfrommer, 2024-12-14:
On Humble (Ubuntu 22.04) the apt package version says 3.0.0, which corresponds to tag 3.0.0 on the ros2-release branch. So checking out tag 3.0.0 on ros2-release should get you the exact sources used for humble.

On Jazzy though it looks to me as we are still on apt package 2.0.20 (not synced yet), so for that you need to check out tag 2.0.20 from ros2-release.

PeteBlackerThe3rd, 2024-12-17:
Quick Update.

I've built the 2.0.20 tag on Ubuntu 20.04 and it still fails to configure the camera, unlike the Debian package version. Patching in the binary and shared object to the installed locations results in the node still failing to configure the camera. I'm not sure what the difference is but the binary and library built are significantly larger than the Debian packaged ones:

```
/opt/ros/jazzy/lib$ ls libcamera* -l
-rw-r--r-- 1 root root   135176 Oct 18 11:17 libcamera_calibration_parsers.so
-rw-r--r-- 1 root root 15913968 Dec 17 09:52 libcamera_driver.so
-rw-r--r-- 1 root root   798952 Dec 15 23:10 libcamera_driver.so.old
-rw-r--r-- 1 root root    97728 Oct 18 11:26 libcamera_info_manager.so
```

```
/opt/ros/jazzy/lib/spinnaker_camera_driver$ ls -l
total 824
-rwxr-xr-x 1 root root 813416 Dec 17 09:52 camera_driver_node
-rwxr-xr-x 1 root root  22880 Oct 18 11:30 camera_driver_node.old
-rwxr-xr-x 1 root root   2313 Jun 20 13:42 linux_setup_flir
```

We're currently building a native Ubuntu 22.04 machine to see if that resolves the issue.

PeteBlackerThe3rd, 2024-12-17:
Another Update.

The node works first time when installed on a native Ubuntu 22.04 machine with the same hardware. So whatever the issue is it's something to do with WSL2. However spinview works fine under WSL2 so whatever the issue it must be solvable. 

Interestingly the Spinnaker SDK examples build for Ubuntu 22.04 under WSL2 also fail. This will probably be my last update since we've found a working solution. I hope this information comes in useful at some point.

closed 2025-01-29 by berndpfrommer