ros-drivers/flir_camera_driver #229: [Humble] FLIR GigE camera with USB camera on a different driver

KrzysztofDubel, 2025-02-03:
I've got two cameras connected with ROS 2 Humble drivers:
- FLIR A50 over GigE with `spinnaker_camera_driver`
- ORBBEC Astra 2 over USB 3.0 with `OrbbecSDK_ROS2`

Both cameras work separately without issues.

1. When I start the FLIR driver first and ORBBEC second, the ORBBEC camera doesn't connect.
```
[component_container-1] [02/03 17:15:50.218340][error][28454][UsbEnumeratorLibusb.cpp:162] Failed to get string descriptor: error=Operation timed out
[component_container-1] [02/03 17:15:50.229070][error][28454][UsbEnumeratorLibusb.cpp:162] Failed to get string descriptor: error=Input/Output Error
```

2. When I start the ORBBEC driver first and FLIR second, the ORBBEC camera disconnects.
It's throwing errors like "bad magic" and "setXu failed"
```
[component_container-1] [02/03 17:15:50.386397][error][28601][ObLibuvcDevicePort.cpp:280] setXu failed, error code=-1
[component_container-1] [02/03 17:15:50.386569][warning][28601][ObException.hpp:48] Request failed, statusCode: 5, msg: Device response with bad magic , magic=0x0, expectOpCode=0x4252
```

I've seen the multiple camera problem in Spinnaker's "Known issues". Is my problem connected to that? Any solutions or workarounds to solve it?

berndpfrommer, 2025-02-05:
Sorry, this kind of problems are very hard to support. Do you see the same behavior with SpinView, i.e. the ORBBEC camera disconnects when you start SpinView?
Suggestion: make sure that USB power is not the issue, i.e. use a USB hub with external power supply, or try the same on a different computer which may deliver more power to its USB devices.
If it's a software thing, that will be harder to address. Is there a way to just pass through a single USB, something along 
 [these lines](https://forums.virtualbox.org/viewtopic.php?t=92074)? I don't know, but that's the only thing I can think about now.

KrzysztofDubel, 2025-02-05:
It actually does work when I run FLIR acquisition through SpinView and the ORBBEC camera still through its ros2 driver.

berndpfrommer, 2025-02-05:
OK, then it must have to do with the way the ROS2 driver initializes the cameras.
To track this time, I would build the driver from source, then, starting from the ``start()`` routine in [camera.cpp](
https://github.com/ros-drivers/flir_camera_driver/blob/832bb95d4fb5eb7adf4ec5575ec648e356df7774/spinnaker_camera_driver/src/camera.cpp) successively comment out more parts of the startup routine until the ORBBEC driver no longer quits. You should be able to track it down to the exact Spinnaker SDK call in the wrapper implementation file that causes the ORBBEC to quit. Then we can think about how to avoid that.

KrzysztofDubel, 2025-02-05:
Thanks!

I will definitely try that.

I tried running `spinnaker_camera_driver` and `orbbecsdk` with another orbbec camera - Femto Bolt. This one resets as soon as spinnaker connects to the FLIR camera but then it gets back up again and is working.

Why does the `spinnaker_camera_driver` reset the other connection?

berndpfrommer, 2025-02-05:
> Why does the `spinnaker_camera_driver` reset the other connection?
I couldn't tell you. I looked over all calls in [spinnaker_wrapper_impl.cpp](https://github.com/ros-drivers/flir_camera_driver/blob/832bb95d4fb5eb7adf4ec5575ec648e356df7774/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp) and don't see anything dangerous being done, like resetting something. And I have other cameras (event based cameras) that keep working just fine when I start the FLIR cameras. No other USB devices are affected either. If this was a common case I would have noticed it and there'd be a mention in the README. So far the only known problem is interference between multiple FLIR cameras.
The fact that SpinView does not screw up the ORBBEC camera throws me a curve ball. If it weren't for that I'd ask you to look into USB resources (kernel memory settings), and if maybe they are insufficient when 2 cameras are running.

KrzysztofDubel, 2025-02-06:
Are there any significant differences in how the connection to the camera is established between `SpinView` and `spinnaker_camera_driver`?

Is the transport layer the same?

berndpfrommer, 2025-02-06:
From what I understand SpinView is using the Spinnaker SDK underneath, but I don't think SpinView is open source, so I cannot say for sure.

berndpfrommer, 2025-06-05:
Did you ever find a resolution to this problem?