ros-drivers/flir_camera_driver #230: How to publish compressed topic data in ROS2

pencilcas, 2025-02-26:
How to publish compressed topic data in ROS2?
I want to change the image format to a compressed one，Where should I make the modification？

berndpfrommer, 2025-02-26:
The ROS2 driver publishes the images through the standard ROS2 image transport. When you do a ``ros2 topic list`` you should see the compressed image topic.
If you don't see it, it could be because you have not installed the image transport plugin. You can do that like so:
```
sudo apt install ros-${ROS_DISTRO}-compressed-image-transport
```
There are other plugins as well (like theora etc).

pencilcas, 2025-02-26:
> The ROS2 driver publishes the images through the standard ROS2 image transport. When you do a `ros2 topic list` you should see the compressed image topic. If you don't see it, it could be because [...]

this is useful and thank you very much for your reply.

closed 2025-02-26 by pencilcas