ros-drivers/flir_camera_driver #240: Acquisition frame rate drops after running ROS2 node

kavishmshah, 2025-05-06:
Hi, 

I am using a Blackfly_S gigE camera (BFS-U3-20S4C-C: 2.0 MP, 175 FPS, Sony IMX422 Color) and running it with this ROS2 package. The issue I see is with respect to the acquisition frame rate which reduced to half after running the ROS2 node. Below are the steps which I tried and was able to repeat the issue. The video attached is the screencast of the steps below:

**1. **Connected the camera --> opened Spinview = 54 FPS video stream
2. Kept camera connected --> Ran the ROS2 package = 26.67 FPS video stream**** 
with warning log " param frame_rate Spinnaker: GenICam::OutOfRangeEception= Value 40.000000 must be smaller than or equal 25.873647. : outOfRangeException thrown in node "_AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 85) [-2002]"
**3. Kept camera connected --> Opened Spinview again = 26 FPS video stream
4. Disconnected camera and connected it back --> Opened Spinview = 54 FPS**

I'm attaching the video along with the ROS2 driver configs. Anyone facing this issue or was able to solve something similar could suggest would be very helpful.
Note: I saw the same issue on multiple devices.

Thanks!

**Video Link:**
https://drive.google.com/file/d/1ZYaj4vEBzSIyeeMlsWIBL03Nv-KSyl0R/view?usp=sharing

kavishmshah, 2025-05-06:
[image]

I tried what you suggested but I don't see that solve the issue.

Another thing I noticed is, I just connected the camera and ran spinview without running the ROS2 node and I still saw the frame rate to be around 5 FPS with all the default camera settings. Iing if there is an issue in the hardware?

kavishmshah, 2025-05-06:
Below are the camera configurations:

     
            [camera_driver_node-1] [ERROR] [1746541122.684439018] 
            [camera_calibration_parsers]: Unable to open camera calibration file 
            [/root/.ros/camera_info/flir_camera.yaml]
            [camera_driver_node-1] [WARN] [1746541122.684455448] [flir_camera]: Camera calibration file /root/.ros/camera_info/flir_camera.yaml not found
            [camera_driver_node-1] [INFO] [1746541124.870799356] [23604040]: using spinnaker lib version: 3.1.0.79
            [camera_driver_node-1] [INFO] [1746541124.871423737] [23604040]: found camera with serial number: 23604040
            [camera_driver_node-1] [INFO] [1746541125.363492122] [Spinnaker Wrapper]: Initialized camera [serial: 23604040] from: [GEV Interface 3]
            [camera_driver_node-1] [INFO] [1746541125.363709029] [23604040]: setting ImageFormatControl/PixelFormat to: BGR8
            [camera_driver_node-1] [INFO] [1746541125.374963689] [23604040]: setting Imag eFormatControl/BinningHorizontal to: 1
            [camera_driver_node-1] [INFO] [1746541125.392891533] [23604040]: setting ImageFormatControl/BinningVertical to: 1
           [camera_driver_node-1] [INFO] [1746541125.410777092] [23604040]: setting AnalogControl/GainAuto to: Off
           [camera_driver_node-1] [INFO] [1746541125.420563452] [23604040]: setting TransportLayerControl/GigEVision/GevSCPSPacketSize to: 9000
           [camera_driver_node-1] [INFO] [1746541125.429855957] [23604040]: setting DigitalIOControl/LineSelector to: Line2
           [camera_driver_node-1] [INFO] [1746541125.433908201] [23604040]: setting AcquisitionControl/ExposureAuto to: Off
           [camera_driver_node-1] [INFO] [1746541125.443067858] [23604040]: setting AcquisitionControl/AcquisitionFrameRateEnable to: 1
           [camera_driver_node-1] [INFO] [1746541125.448677878] [23604040]: setting AcquisitionControl/AcquisitionFrameRate to: 40
           [camera_driver_node-1] [WARN] [1746541125.454310307] [23604040]: param frame_rate Spinnaker: GenICam::OutOfRangeException= Value 40.000000 must be smaller than or equal 2.064849. : OutOfRangeException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 85) [-2002]
           [camera_driver_node-1] [INFO] [1746541125.454467407] [23604040]: setting AcquisitionControl/TriggerMode to: Off
            [camera_driver_node-1] [INFO] [1746541125.459531736] [23604040]: setting AcquisitionControl/TriggerSource to: Line3
            [camera_driver_node-1] [INFO] [1746541125.465723771] [23604040]: setting AcquisitionControl/TriggerActivation to: RisingEdge
            [camera_driver_node-1] [INFO] [1746541125.471734015] [23604040]: setting ChunkDataControl/ChunkModeActive to: 1
            [camera_driver_node-1] [INFO] [1746541125.478811467] [23604040]: setting ChunkDataControl/ChunkSelector to: FrameID
            [camera_driver_node-1] [INFO] [1746541125.483796021] [23604040]: setting ChunkDataControl/ChunkEnable to: 1
            [camera_driver_node-1] [INFO] [1746541125.489240146] [23604040]: setting ChunkDataControl/ChunkSelector to: ExposureTime
            [camera_driver_node-1] [INFO] [1746541125.490687030] [23604040]: setting ChunkDataControl/ChunkEnable to: 1
            [camera_driver_node-1] [INFO] [1746541125.495183040] [23604040]: setting ChunkDataControl/ChunkSelector to: Gain
            [camera_driver_node-1] [INFO] [1746541125.496792158] [23604040]: setting ChunkDataControl/ChunkEnable to: 1
            [camera_driver_node-1] [INFO] [1746541125.501180010] [23604040]: setting ChunkDataControl/ChunkSelector to: Timestamp
            [camera_driver_node-1] [INFO] [1746541125.502705485] [23604040]: setting ChunkDataControl/ChunkEnable to: 1
            [camera_driver_node-1] [INFO] [1746541125.697862825] [23604040]: camera has pixel format: BGR8
            [camera_driver_node-1] [INFO] [1746541125.697953344] [flir_camera]: camera_driver_node started up!
            [camera_driver_node-1] [INFO] [1746541127.679600772] [23604040]: rate [Hz] in   2.06 out   0.00 drop   0%
             [camera_driver_node-1] [INFO] [1746541132.679407552] [23604040]: rate [Hz] in   2.06 out   0.00 drop   0%
             [camera_driver_node-1] [INFO] [1746541137.679569647] [23604040]: rate [Hz] in   2.06 out   0.00 drop   0%
             [camera_driver_node-1] [INFO] [1746541142.679560969] [23604040]: rate [Hz] in   2.06 out   0.00 drop   0%
             [camera_driver_node-1] [INFO] [1746541147.679432612] [23604040]: rate [Hz] in   2.06 out   0.00 drop   0%

berndpfrommer, 2025-05-07:
This feedback clearly comes from the camera. It's telling you the maximum frame rate is 2.06, and that's exactly what you get. Possible reasons:
1) You are not getting the full bandwidth from the network. With only 2MP resolution, that's probably not the problem.
2) Not enough light. How does the image look you are getting? If you go into SpinView, can you see what the exposure time is? Is the exposure time close to 1/frame_rate?

You can set the exposure time explicitly to something every low, that should eliminate 2) as a possible cause.

kavishmshah, 2025-05-07:
1. Yup, I dont think that should be the problem because it works just fine sometimes while it doesn't otherwise. I wasn't seeing this issue occur previously and was able to stream upto 55 FPS.
2. My working environment is pretty bright with good amount of natural light including outdoors on sunny days. Exposure time was default set to 14990 (microseconds). I also changed the Exposure mode to 'Off' from 'Continuous' and modified the exposure time to as low as 500. All I could see was the video stream being dark but the FPS didn't change much (remained around 5 FPS).

berndpfrommer, 2025-05-07:
All indications are that some camera settings goes awry which causes the camera to subsequently reduce the frame rate. Check that you have the latest firmware.
You can also try resetting the camera, then storing the known-good settings as e.g. User Settings 1 in the camera, then loading them back with the ros driver. A workaround maybe.

kavishmshah, 2025-05-07:
I haven't updated the firmware since I got the camera which has been long but I did factory reset the camera, but even with the factory reset I saw the low FPS in random.

berndpfrommer, 2025-05-10:
When it goes into this low-fps state try bringing up SpinView and find out what parameter exactly it is that it causing the camera to reduce fps. Is there any way you can get it back to full frame rate without reset?

kavishmshah, 2025-05-13:
So what I tried is the below (using [link](https://www.teledynevisionsolutions.com/support/support-center/application-note/iis/saving-custom-settings-on-flir-machine-vision-cameras/))

1. Open SpinView.
2. Select the camera.
3. In the Features tab, under the camera model name, make whatever camera settings changes you want to keep (for example, change exposure auto to off).
4. In the Features tab, under the camera model name, click User Set Control.
5. Change User Set Selector to User Set 0 (or User Set 1).
6. Click the Execute button beside User Set Save.
7. Change User Set Default to User Set 0 (or User Set 1 depending on step 5).

After doing the above, it seems like I don't see the low-fps state now. (tried it a few times). Do you have an explanation of why this might've worked?

Also, when the config are set through the ROS2 launch file, do the parameters hold up the value, if yes, does spinview overwrite them when changes are made in spinview?

Thanks!

berndpfrommer, 2025-05-14:
If the ros driver sets bad parameters on the cam, bringing up spinview won't reset them. 
The only explanation I have is that you still were/are loading usersetX when the ros driver starts. Your last log didn't show that however so that doesn't make sense either. Do some more testing and if the issue doesn't come back please close it out.

berndpfrommer, 2025-05-14:
And yes, both changes made with the ros driver or spin view survive replugging.

kavishmshah, 2025-05-14:
and do you know how the usersetX parameters get loaded in or where the parameters for usersetX are stored? This is because I don't recollected changing any of those and keep in mind when any of this changes in the future.

berndpfrommer, 2025-05-15:
If you use the default launch file the UserSet does not get loaded, because I commented out [here](https://github.com/ros-drivers/flir_camera_driver/blob/2884501346a3404489226f916e6dc7a48dcba977/spinnaker_camera_driver/launch/driver_node.launch.py#L37). There is no logic in the driver that would suddenly load the user parameters unless you command it to do so by setting the right ros parameters (the ones that are commented out in the example launch file).

kavishmshah, 2025-05-15:
Gotcha. 
So is the UserSet that gets loaded in stored on a chip on the camera and those parameters get loaded in when commanded to do so? On spinview I can see UserSet 0 and UserSet1 can be used.

kavishmshah, 2025-05-15:
Perfect, thanks! I'll do some testing and close the issue if I don't see any other problems.

kavishmshah, 2025-06-03:
I figured out that either:

1. the userset parameters needed to  be updated as @berndpfrommer suggested.
2. Also, one of my POE cables seemed faulty, which may have caused this.

Closing this issue.