ros-drivers/flir_camera_driver #242: [Humble][flir_camera]: startup failed!

youngjoo842, 2025-05-12:
Hello.
I'm using BFS-U3-31S4C-C camera, and running with ros2 humble package.

When I check lsusb, I can see the camera is connected.
Bus 002 Device 003: ID 1e10:4000 Point Grey Research, Inc. Blackfly S BFS-U3-31S4C

I downloaded Spinnaker 4.2 for Linux.
When I launch driver_node, it cannot find camera.

ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'24135500'"
[INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2025-05-12-18-15-11-840790-nvidia-Vector-17-HX-A13VHG-16709
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [16720]
[camera_driver_node-1] [INFO] [1747041311.895136830] [24135540]: reading ros parameters for camera with serial: 24135540
[camera_driver_node-1] [INFO] [1747041311.895190449] [24135540]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1747041311.895194933] [24135540]: not using IEEE 1588 (PTP)!
[camera_driver_node-1] [INFO] [1747041311.895276366] [24135540]: parameter definitions file: /home/nvidia/temp_ws/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[camera_driver_node-1] [INFO] [1747041311.896981782] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1747041311.897008509] [flir_camera]: camera calibration URL: file:///home/nvidia/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [ERROR] [1747041311.897031738] [camera_calibration_parsers]: Unable to open camera calibration file [/home/nvidia/.ros/camera_info/flir_camera.yaml]
[camera_driver_node-1] [WARN] [1747041311.897035998] [flir_camera]: Camera calibration file /home/nvidia/.ros/camera_info/flir_camera.yaml not found
[camera_driver_node-1] [INFO] [1747041313.120455789] [24135540]: using spinnaker lib version: 4.2.0.46
[camera_driver_node-1] [WARN] [1747041313.120527012] [24135540]: no camera found with serial: 24135500 on try # 1
[camera_driver_node-1] [WARN] [1747041314.121575043] [24135540]: no camera found with serial: 24135500 on try # 2
[camera_driver_node-1] [WARN] [1747041315.122522482] [24135540]: no camera found with serial: 24135500 on try # 3
[camera_driver_node-1] [WARN] [1747041316.123553036] [24135540]: no camera found with serial: 24135500 on try # 4
[camera_driver_node-1] [WARN] [1747041317.124588346] [24135540]: no camera found with serial: 24135500 on try # 5
[camera_driver_node-1] [ERROR] [1747041318.124917931] [24135540]: giving up, camera 24135540 not found!
[camera_driver_node-1] [ERROR] [1747041318.125022911] [flir_camera]: startup failed!
[camera_driver_node-1] [INFO] [1747041318.125115697] [flir_camera]: camera_driver_node started up!
[camera_driver_node-1] [INFO] [1747041318.125784266] [24135540]: rate [Hz] in   0.00 out   0.00 drop   0
[camera_driver_node-1] [INFO] [1747041321.895437884] [24135540]: rate [Hz] in   0.00 out   0.00 drop   0
[camera_driver_node-1] [INFO] [1747041326.895447537] [24135540]: rate [Hz] in   0.00 out   0.00 drop   0

Do I need to do any additional setup after building?

berndpfrommer, 2025-05-12:
Does spinview work?

MohamadAshrafSalama, 2025-05-12:
i have the same problem and yes  spinview works

berndpfrommer, 2025-05-13:
If you install from the apt package, does that work? It uses a different version of the Spinnaker library.

youngjoo842, 2025-05-13:
No. Spinview doesn't work.

Info: (null) ((null):0, Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.)
QIODevice::write (QFile, "SpinView_QT_log"): device not open

[image]

berndpfrommer, 2025-05-13:
Then it's a FLIR problem, not a ros driver problem. I have had pretty good responses from FLIR tech support, give it a shot.
Are you using any exotic hardware, or WSL? A docker container, a VM? Could also be a bad USB cable. The cameras are very finicky. I only use original flir cables.
Lastly, check if permissions are an issue. Run strace -f spinview >& foo.txt and grep foo.txt for EACCESS, EPERM etc. Anything obvious there?

youngjoo842, 2025-05-13:
I use laptop with Ubuntu 22.04 installed. I changed extra cable and it dosen't work too. But it's not original flir cables. Maybe it's the problem. Thank you. I will ask FLIR.

youngjoo842, 2025-05-14:
When I use `sudo /opt/spinnaker/bin/SpinView_QT`, I can get the Image from the camera. But, I launch driver_node.launch.py, it dosen't work.

berndpfrommer, 2025-05-14:
Does your user belong to the "flirimaging" group? You had the choice to set the up during Spinnaker install.
Type "groups" and see if "flirimaging" is there. If yes, then reboot once and see if you can use SpinView without sudo. You need the access permissions right for the ros driver to work.

youngjoo842, 2025-05-15:
After reinstall Spinnaker, it works. Thank you!

closed 2025-05-15 by youngjoo842