ros-drivers/flir_camera_driver #246: Driver crashes if camera disconnects and doesn't recover after reconnect

sausais, 2025-06-12:
Description
When using the spinnaker_camera_driver, the driver crashes entirely if the FLIR camera is disconnected even for a very short moment (e.g. 100 ms). 
After this short disconnect, the driver throws a Spinnaker::Exception, shuts down, and does not reconnect or recover when the camera is plugged back in. From that point on, the camera stays dead until the entire ROS node is manually restarted.

Expected Behavior

Automatically reconnect and resume streaming when the camera is available again (like SpinView does)

berndpfrommer, 2025-06-18:
I never tried that, so no surprise it doesn't work. What camera do you use (model)? Particularly interesting if you use USB3 vs GigE.

GreatAlexander, 2025-07-31:
Hi @berndpfrommer , for reference we have the same issue (USB3, Model CM3-U3-50S5C).
Our cameras are on a mobile platform (car) and vibrations sometimes cause the cameras to temporarily disconnect, making the ROS driver crash. We have taped the USB3 connectors with conductive tape to minimise this issue.

jclinton830, 2025-10-02:
@berndpfrommer have you made any progress on this front? We are also facing a similar problem. Sometimes, the camera composable node fails, and we need a method to self-heal and bring the composable node back up. Happy to help with the development.

berndpfrommer, 2025-10-05:
Sorry folks, I was working on other stuff. But I just managed to get PR #253 over the finish line, so I should be able to start working on this on Wednesday.