ros-drivers/flir_camera_driver #247: [JAZZY] Stereo Image Frame Rate Drop - When using with foxglove bridge and ROS2 Bag Record

jclinton830, 2025-06-18:
**Describe the bug**
HI, 

NOTE - RMW_ZENOH_CPP is used in this setup. 

I can run the [primary_secondary.launch.py](https://github.com/ros-drivers/flir_camera_driver/blob/61d52a5e1d618631c7f291592ca887a16c199090/spinnaker_synchronized_camera_driver/launch/primary_secondary.launch.py) to start both my blackfly_s cameras, and the acquisition starts and goes at a steady 24 Hz. See log below 

```
/ros_ws$ ros2 launch groundhog_bringup stereo.launch.py 
[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2025-06-18-03-00-05-160328-groundhog-5296
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [5316]
[component_container-1] [INFO] [1750215605.708180609] [cam_sync_container]: Load Library: /opt/ros/jazzy/lib/libsynchronized_camera_driver.so
[component_container-1] [INFO] [1750215605.749901179] [cam_sync_container]: Found class: rclcpp_components::NodeFactoryTemplate<spinnaker_camera_driver::CameraDriver>
[component_container-1] [INFO] [1750215605.749940120] [cam_sync_container]: Found class: rclcpp_components::NodeFactoryTemplate<spinnaker_synchronized_camera_driver::SynchronizedCameraDriver>
[component_container-1] [INFO] [1750215605.749948441] [cam_sync_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<spinnaker_synchronized_camera_driver::SynchronizedCameraDriver>
[component_container-1] [INFO] [1750215605.752970638] [cam_sync]: created exposure controller: cam0.exposure_controller
[component_container-1] [INFO] [1750215605.753171518] [cam_sync]: created exposure controller: cam1.exposure_controller
[component_container-1] [INFO] [1750215605.753827095] [cam0]: parameter definitions file: /opt/ros/jazzy/share/spinnaker_camera_driver/config/blackfly_s.yaml
[component_container-1] [INFO] [1750215605.759686681] [cam_sync]: camera calibration URL: file:///home/ros/.ros/camera_info/21070683.yaml
[component_container-1] [ERROR] [1750215605.759792919] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ros/.ros/camera_info/21070683.yaml]
[component_container-1] [WARN] [1750215605.759837256] [cam_sync]: Camera calibration file /home/ros/.ros/camera_info/21070683.yaml not found
[component_container-1] [INFO] [1750215606.958202429] [cam0]: using spinnaker lib version: 4.0.0.116
[component_container-1] [INFO] [1750215606.958451836] [cam0]: found camera with serial number: 21070683
[component_container-1] [INFO] [1750215607.276358446] [Spinnaker Wrapper]: Initialized camera [serial: 21070683] from: [USB Interface 0]
[component_container-1] [INFO] [1750215607.339710693] [cam0]: camera has pixel format: BGR8
[component_container-1] [INFO] [1750215607.340220670] [cam1]: parameter definitions file: /opt/ros/jazzy/share/spinnaker_camera_driver/config/blackfly_s.yaml
[component_container-1] [INFO] [1750215607.341856312] [cam_sync]: camera calibration URL: file:///home/ros/.ros/camera_info/21070659.yaml
[component_container-1] [ERROR] [1750215607.341920897] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ros/.ros/camera_info/21070659.yaml]
[component_container-1] [WARN] [1750215607.341953125] [cam_sync]: Camera calibration file /home/ros/.ros/camera_info/21070659.yaml not found
[component_container-1] [INFO] [1750215607.344941921] [cam1]: using spinnaker lib version: 4.0.0.116
[component_container-1] [INFO] [1750215607.345099053] [cam1]: found camera with serial number: 21070659
[component_container-1] [INFO] [1750215607.384191740] [cam0]: bright 1 at time/gain: [2004 29.8965] new: [2000 29.8965]
[component_container-1] [INFO] [1750215607.464843962] [cam0]: bright 1 at time/gain: [2000 29.8965] new: [2000 29.9]
[component_container-1] [INFO] [1750215607.628820107] [cam_sync]: frequency estimator initialized with 24.488 Hz
[component_container-1] [INFO] [1750215607.686998501] [Spinnaker Wrapper]: Initialized camera [serial: 21070659] from: [USB Interface 0]
[component_container-1] [INFO] [1750215607.753177484] [cam1]: camera has pixel format: BGR8
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/cam_sync' in container '/cam_sync_container'
[component_container-1] [INFO] [1750215607.847680660] [cam1]: bright 1 at time/gain: [2004 29.8965] new: [2000 29.9]
[component_container-1] [INFO] [1750215612.753338655] [cam_sync]: ------ frequency:     24.340 Hz
[component_container-1] [INFO] [1750215612.753406403] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1750215612.753625065] [cam_sync]: cam0        0    0     0.06%     0.78%
[component_container-1] [INFO] [1750215612.753643839] [cam_sync]: cam1        0    0     0.10%     0.74%
[component_container-1] [INFO] [1750215617.753329376] [cam_sync]: ------ frequency:     24.343 Hz
[component_container-1] [INFO] [1750215617.753399728] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1750215617.753415809] [cam_sync]: cam0        0    0    -0.03%     0.20%
[component_container-1] [INFO] [1750215617.753432048] [cam_sync]: cam1        0    0     0.03%     0.16%
[component_container-1] [INFO] [1750215622.753339904] [cam_sync]: ------ frequency:     24.346 Hz
[component_container-1] [INFO] [1750215622.753415318] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1750215622.753434690] [cam_sync]: cam0        0    0    -0.06%     0.23%
[component_container-1] [INFO] [1750215622.753450951] [cam_sync]: cam1        0    0     0.05%     0.26%
[component_container-1] [INFO] [1750215627.753355427] [cam_sync]: ------ frequency:     24.346 Hz
[component_container-1] [INFO] [1750215627.753421354] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1750215627.753437482] [cam_sync]: cam0        0    0    -0.08%     0.12%
[component_container-1] [INFO] [1750215627.753453517] [cam_sync]: cam1        0    0     0.08%     0.20%
[component_container-1] [INFO] [1750215632.753291914] [cam_sync]: ------ frequency:     24.348 Hz
[component_container-1] [INFO] [1750215632.753376242] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1750215632.753392589] [cam_sync]: cam0        0    0    -0.11%     0.17%
[component_container-1] [INFO] [1750215632.753411568] [cam_sync]: cam1        0    0     0.09%     0.26%
```

```
ros2 topic hz /cam_sync/cam0/image_raw
average rate: 25.001
	min: 0.014s max: 0.062s std dev: 0.00784s window: 26
average rate: 24.933
	min: 0.014s max: 0.064s std dev: 0.00751s window: 51
average rate: 24.736
	min: 0.014s max: 0.064s std dev: 0.00642s window: 76
average rate: 24.628
[... 288 lines cut ...]
$ ros2 topic hz /cam_sync/cam0/image_raw
average rate: 1.081
	min: 0.912s max: 0.950s std dev: 0.01751s window: 3
average rate: 1.054
	min: 0.912s max: 1.015s std dev: 0.03739s window: 5
average rate: 1.026
	min: 0.912s max: 1.104s std dev: 0.06732s window: 6
average rate: 1.007
	min: 0.912s max: 1.106s std dev: 0.07747s window: 7
average rate: 1.015
	min: 0.912s max: 1.106s std dev: 0.07003s window: 9
average rate: 1.027
	min: 0.902s max: 1.106s std dev: 0.06836s window: 11
average rate: 1.019
	min: 0.902s max: 1.106s std dev: 0.07017s window: 12
average rate: 1.017

```

Any help in troubleshooting why this is happening would be great. 

**System details**
- operating system/version: Docker container with 24.04
- ROS version: ROS2 Jazzy
- complete FLIR camera model: Blackfly S BFS-U3-50S5C

berndpfrommer, 2025-06-18:
On a USB3 connection you really shouldn't see frame gaps.
The only thing that I can spot is that you are subscribing to all topics. The image transport has to compress then, which is CPU intensive, and may bog down your host to the point that the camera driver no longer has the CPU resources to pick up the incoming frames. What does "top" say? Are you maxxed out on CPU? How about *not* recording the images three times: raw, compressed and theora compressed? That should do a whole lot of good to your CPU usage.
Please let me know how you make out.

jclinton830, 2025-06-18:
@berndpfrommer, you are right. The CPU was being exhausted too much. After removing all the extra image transports, I am getting a steady 24Hz on the ROS2 side, even with Foxglove Studio. 

However, the composable recorder is still causing a drop of about 10 Hz. 

I think I can move the data logging to our logging computer with the same ROS_DOMAIN_ID and improve the CPU overhead to solve the issue. I will try this tomorrow and let you know.

jclinton830, 2025-10-16:
After upgrading to a better computer instead of using something almost a decade old, this has been resolved.