ros-drivers/flir_camera_driver #252: Delayed timestamps

danicannt, 2025-09-18:
**Is this behavior also observed when using SpinVIew?**
No

**Describe the bug**
Camera messages are being published with an approximately 10-second delay compared to other sensors (e.g., LiDAR). This causes synchronization issues in data processing. The delay seems related to how Camera::doPublish() assigns timestamps, especially when synchronizer_ or IEEE1588/PTP is not used.

**Useful logging output**
Example of published timestamps:
`Camera:
stamp:
  sec: 1758184496
  nanosec: 324700171
frame_id: cam_1

LiDAR sensor:
stamp:
  sec: 1758184505
  nanosec: 580716895
frame_id: ouster_data_frame
`
Relevant snippet from the camera publishing code:
`void Camera::doPublish(const ImageConstPtr & im)
{
  rclcpp::Time t;
  if (synchronizer_) {
    uint64_t t_64;
    bool haveTime = synchronizer_->getTimeStamp(
      im->time_, im->imageTime_, im->frameId_, im->numIncomplete_, &t_64);
    t = rclcpp::Time(t_64, RCL_SYSTEM_TIME);
    if (!haveTime) {
      if (firstSynchronizedFrame_) {
        firstSynchronizedFrame_ = false;
      } else {
        LOG_WARN("cannot get time stamp for frame, dropping it");
      }
      return;
    }
  } else {
    if (useIEEE1588_) {
      t = rclcpp::Time(im->imageTime_, RCL_SYSTEM_TIME);
      ptpOffset_ = (static_cast<int64_t>(im->time_) - static_cast<int64_t>(im->imageTime_)) * 1e-9;
    } else {
      t = adjustTimeStamp_ ? getAdjustedTimeStamp(im->time_, im->imageTime_)
                           : rclcpp::Time(im->time_);
    }
  }
}
`

**System details**
operating system/version: Ubuntu 22.04
ROS version: ROS 2 Humble
complete FLIR camera model: BFS-U3-04S2C-C

berndpfrommer, 2025-09-18:
- are you running with adjustTimeStamp_ == true?
- if you ``ros2 bag record`` the images *on the same host where the camera driver is running*, do the header.stamp time stamps agree with the rosbag recording time stamp? You can check that easily with e.g. rqt_bag (show raw data will let you examine the header stamp, while the recording time stamp is displayed at the bottom of the GUI).

berndpfrommer, 2025-10-16:
Closing for lack of response.

closed 2025-10-16 by berndpfrommer